vectr-ucla/direct_lidar_odometry #1: rosbag seems inaccessible

libing64, 2021-12-02:
The rosbag seems inaccessible, would you like to share another link for downloading?

kennyjchen, 2021-12-02:
Hi @libing64 -- could you please post a screenshot or elaborate on the symptoms? The download link is still working for me. FYI the file is hosted on Box.

narutojxl, 2021-12-03:
With a vpn in china mainland can  download, otherwise can not.

libing64, 2021-12-03:
> Hi @libing64 -- could you please post a screenshot or elaborate on the symptoms? The download link is still working for me. FYI the file is hosted on Box.

[image]
[image]

libing64, 2021-12-03:
> With a vpn in china mainland can download, otherwise can not.

@narutojxl My network has access to foreign networks and I could use goolge/youtube normally.

narutojxl, 2021-12-03:
It's werid, i don't know whether it is Box internet stability. I can download now, but sometimes can not. Maybe google drive is stable ?
[image]

libing64, 2021-12-04:
The bag could be download with my home network with a vpn, great thanks for your sharing.

closed 2021-12-04 by libing64

2653888596, 2022-10-06:
> The bag could be download with my home network and a vpn, great thanks for your sharing.

Hello, could you please put this rosbag onto BaiDuYun? I will very appreciate. Thank U