vectr-ucla/direct_lidar_odometry #2: how to set a different initialpose?

numb0824, 2021-12-31:
Hi,
  thanks for your team's great work.And i have tested it with my own bag, the result is very close for my ground truth.
  There is a request that i want to set the initial pose ,such as (100,100,0,0,0,1,0.026). But it didn't work,the "robot/dlo/odom" has always begin with (0,0,0,0,0,0,1). I just changed the code in "odom.cc",which is:
"
  this->odom.pose.pose.position.x = 3972.02;
  this->odom.pose.pose.position.y = 3497.76;
  this->odom.pose.pose.position.z = 0;
  this->odom.pose.pose.orientation.w = 0.026;
  this->odom.pose.pose.orientation.x = 0.;
  this->odom.pose.pose.orientation.y = 0.;
  this->odom.pose.pose.orientation.z = 1;
  this->odom.pose.covariance = {0.};
  
/*
  this->odom.pose.pose.position.x = 0.;
  this->odom.pose.pose.position.y = 0.;
  this->odom.pose.pose.position.z = 0.;
  this->odom.pose.pose.orientation.w = 1.;
  this->odom.pose.pose.orientation.x = 0.;
  this->odom.pose.pose.orientation.y = 0.;
  this->odom.pose.pose.orientation.z = 0.;
  this->odom.pose.covariance = {0.};
*/
  this->T = Eigen::Matrix4f::Identity();
  this->T_s2s = Eigen::Matrix4f::Identity();
  this->T_s2s_prev = Eigen::Matrix4f::Identity();

  this->pose_s2s = Eigen::Vector3f(0., 0., 0.);
  this->rotq_s2s = Eigen::Quaternionf(1., 0., 0., 0.);

  //this->pose = Eigen::Vector3f(0., 0., 0.);
  this->pose = Eigen::Vector3f(3972.02, 3497.76, 0.);
  //this->rotq = Eigen::Quaternionf(1., 0., 0., 0.);
  this->rotq = Eigen::Quaternionf(0.026, 0., 0., 0.);
"
I would really appreciate it if someone could help me.

kennyjchen, 2022-01-04:
Hi @numb0824 -- thanks for your interest in our work. We've added a feature to initialize DLO with a known initial pose in [`v1.1.0`](https://github.com/vectr-ucla/direct_lidar_odometry/releases/tag/v1.1.0). Please set the `initialPose` parameter in `cfg/params.yaml` to `true` with your desired initial pose. Note that this will disable the gravity alignment procedure.

closed 2022-01-11 by kennyjchen

numb0824, 2022-01-17:
Thanks a lot, it really help me!