vectr-ucla/direct_lidar_odometry #4: the transformation between your lidar and imu?

nonlinear1, 2022-01-10:
I use a vlp16 lidar and a microstran imu(3DM-GX5-25) to run your program. But I failed. I think my transformation between imu and lidar is different from yours. So could you tell your true transformation between imu and lidar. the following pitcure is my configuration:
[image]

kennyjchen, 2022-01-10:
Hi @nonlinear1, could you elaborate a bit on how it's failing? Could you try disabling the IMU by setting ```imu: false``` in ```cfg/dlo.yaml```?

nonlinear1, 2022-01-11:
Thank you for your help! I have set imu: false in cfg/dlo.yaml adopting your suggestion. But I still 
failed . see the following pictures

[image]
[image]

[image]

kennyjchen, 2022-01-11:
Lol yeah that definitely doesn't look right. Does it work with the test bag? I see that you're on v1.0.0; can you try using the latest code on the `master` branch? And can you record me a bag with your own data so I can help debug?

nonlinear1, 2022-01-11:
Thank you very much! I solve the problem.  I change the configuration of my vlp16. Your slam is very good! 
[image]

nonlinear1, 2022-01-11:
I solve this problem. I change my vlp16 configuration and I got the good result

closed 2022-01-11 by nonlinear1

kennyjchen, 2022-01-11:
Nice! Glad to hear that.