vectr-ucla/direct_lidar_odometry #7: getSubmapKeyframes()

narutojxl, 2022-04-11:
Hello authors,
Thanks for sharing this excellent work to let us study firstly :) 
In [getSubmapKeyframes()](https://github.com/vectr-ucla/direct_lidar_odometry/blob/master/src/dlo/odom.cc#L1240) we call computeConvexHull() to calculate the keyframes indice whose position(xyz) consist of convex hull, we then push the indices into `keyframe_convex`. Now let we suppose keyframe_convex's content is  `keyframe_convex = [3,  6,  8,  10, 13].`  We then push the distance between curr scan position to each of keyframe_convex position one by one from begining into `convex_ds.`  Let we suppose convex_ds is `convex_ds =[1.0,  0.2,  1.4,  0.3,  3.0]`.  Then in `pushSubmapIndices()` we calculate submap_kcv_ smallest elements, which represent submap_kcv_ nearest convex keyframes indices from current scan position. Let we suppose  submap_kcv_ = 2. In [pushSubmapIndices](https://github.com/vectr-ucla/direct_lidar_odometry/blob/master/src/dlo/odom.cc#L1230) distance 0.2 and 0.3 will be sorted, so we push indice = 1 and indice = 3 into `submap_kf_idx_curr`, but i think we should push the corresponding indices in `keyframe_convex` namely 6 and 10 into `submap_kf_idx_curr`, which are distance 0.2 and 0.3 correspond keyframe indices, right?  
If i mistake somewhere code please correct me. If I'm right, the same problem also in [pushSubmapIndices() of computeConcaveHull()](https://github.com/vectr-ucla/direct_lidar_odometry/blob/master/src/dlo/odom.cc#L1289).
BTW, [this->pose in L1252](https://github.com/vectr-ucla/direct_lidar_odometry/blob/master/src/dlo/odom.cc#L1252) is last scan position in global frame, why not use the current scan's guess value in global frame, namely `this->T_s2s`, we get from [this->propagateS2S(T_S2S);](https://github.com/vectr-ucla/direct_lidar_odometry/blob/master/src/dlo/odom.cc#L812). 
Thanks for your time and help very much in advance!

kennyjchen, 2022-04-11:
Hi @narutojxl -- thanks for your interest in our work!

Great catch -- the indices should be extracted from the original list of convex/concave keyframes (rather than from the distance vector). Your second point is also correct in that we should be using the most current positional estimate during the search (although in practice this difference is most likely negligible).

Thanks for the keen eye! We've fixed both of these bugs in `v1.3.0` and have credited you accordingly. Let us know if there are any other issues.

narutojxl, 2022-04-12:
Thanks author, close now.