vectr-ucla/direct_lidar_odometry #8: some trivial questions

narutojxl, 2022-04-11:
Hello authors,
I have some trivial questions to look for your help. Thanks for your help.
* [diff rot](https://github.com/vectr-ucla/direct_lidar_odometry/blob/master/src/dlo/odom.cc#L1137) is neither `this->rotq.inverse() * closest_pose_r` nor `closest_pose_r.inverse() * this->rotq`?
* [quaternion to axis angle](https://github.com/vectr-ucla/direct_lidar_odometry/blob/master/src/dlo/odom.cc#L1139), rotation norm is `theta_rad *=2;`
* [this->keyframe_cloud](https://github.com/vectr-ucla/direct_lidar_odometry/blob/master/src/dlo/odom.cc#L1169) is the last keyframe, we should set curr keyframe into gicp_s2s.setInputSource(), before we calculate covariance, namely [this->keyframe_cloud](https://github.com/vectr-ucla/direct_lidar_odometry/blob/master/src/dlo/odom.cc#L1174) update should before we calculate covariance.
* [returned final rotation](https://github.com/vectr-ucla/direct_lidar_odometry/blob/master/src/dlo/odom.cc#L840) should be check whether to flip sign, rather than in  [publishPose()](https://github.com/vectr-ucla/direct_lidar_odometry/blob/master/src/dlo/odom.cc#L341)

kennyjchen, 2022-04-11:
Hi @narutojxl --

- The difference quaternion between two quaternions can be computed by multiplying one with the inverse of the other.
- You are right; thanks for the catch! We've fixed this in `1.3.1`
- Good catch again -- the update should come before the covariance calculation. Fixed in `1.3.1`
- Since quaternions are a double map of SO3, a quaternion `q` and `-q` represent the same rotation so sign flips ultimately won't affect the solution. Sign flips are not preferable for controllers, however, so we chose to keep the check prior to publishing to minimize number of operations in the main loop.

Thanks again very much for your questions and bug catches.

narutojxl, 2022-04-12:
Thanks author fast reply.
For the first question in code, `Eigen::Quaternionf dq = this->rotq * (closest_pose_r.inverse());`, does this `dq` have a actual physical meaning?  Because `this->rotq.inverse() * closest_pose_r` represents curr scan to cloest keyframe rot, `closest_pose_r.inverse() * this->rotq` represents cloest keyframe to curr scan rot, they all have a actual physical meaning.

kennyjchen, 2022-04-12:
Yes, `dq` is the relative rotation which brings `closest_pose_r` to `this->rotq`, such that
```
dq * closest_pose_r = this->rotq
```
or
```
dq = this->rotq * closest_pose_r.inverse()
```

Since we're interested in the magnitude between the current pose and a keyframe, it doesn't matter which "direction" we go since we take the absolute value of this angle (lines `1145, 1149, 1151`).

narutojxl, 2022-04-12:
OK, I see its meaning is a global delta_rot which transforms cloest_pose_r into curr rot.  Thanks very much! My problem is solved, close it now.

closed 2022-04-12 by narutojxl