vectr-ucla/direct_lidar_odometry #11: how to localize on a given pcd map?

srinivasrama, 2022-05-12:
i have a trouble understanding how to localise on a given 3d pcd map, can anyone explain the steps

i have a pcd map which i can load into rviz, and i have lidar-points and imu data, then how to use this package for localization?

kennyjchen, 2022-05-12:
Hi @srinivasrama -- DLO currently does not support localizing within a provided map and assumes no prior knowledge of the environment. However, I don't imagine this would be hard: instead of [detecting keyframes](https://github.com/vectr-ucla/direct_lidar_odometry/blob/master/src/dlo/odom.cc#L678) and [extracting a submap](https://github.com/vectr-ucla/direct_lidar_odometry/blob/master/src/dlo/odom.cc#L825) on each callback, load in your PCD map (in the constructor) and scan match with that on every iteration (your map being the target cloud). Note that you may need to provide the system with an initial pose though. I can take a jab at it when I have some free time, but that might not be for a while. Otherwise, pull requests are always welcome.

srinivasrama, 2022-05-13:
@kennyjchen Thanks

closed 2022-05-13 by srinivasrama

2653888596, 2022-10-21:
@kennyjchen Hello, I admire your work very much, and it works well on the data collected by ourselves. Have you been doing the method of pure localization by using PCD map recently? I look forward to your open source of this part of the code! Thanks!

2653888596, 2022-10-21:
> @kennyjchen Thanks

@srinivasrama Hello, I have encountered the same problem. As I am a beginner, have you solved the whole problem? Would it be convenient for you to communicate with me?

srinivasrama, 2022-10-21:
@2653888596 i used NDT from autoware for the localization

2653888596, 2022-10-21:
@srinivasrama I'm very curious about how you did it. Can you share your code? Thank you very much. If you can, you can give me a link or send it to my email address z2653888596@gmail.com.