vectr-ucla/direct_lidar_odometry #12: Covariances Calculation in updatekeyframe

qweesiic, 2022-05-20:
great job!

i found covariances was calculated in function updatekeyframe();
and  i am not sure if  
[image]
was  executed after function updatekeyframe()， the time of covariances calculation in function updatekeyframe() maybe saved？

kennyjchen, 2022-05-20:
Hi @qweesiic --

We calculate covariances twice for each acquired scan: 
- In `base_link` frame in `getNextPose()` by calling `align()` [here](https://github.com/vectr-ucla/direct_lidar_odometry/blob/master/src/dlo/odom.cc#L803), which is an inherited `pcl::Registration` [function](https://pointclouds.org/documentation/registration_8hpp_source.html#l00173) that calls nano_gicp's templated [`computeTransformation()`](https://github.com/vectr-ucla/direct_lidar_odometry/blob/master/include/nano_gicp/impl/nano_gicp_impl.hpp#L162) (which calculates covariances if none are available)
- In `odom` frame in `updateKeyframes()` [here](https://github.com/vectr-ucla/direct_lidar_odometry/blob/master/src/dlo/odom.cc#L1173) by explicitly calling nano_gicp's [`calculateSourceCovariances()`](https://github.com/vectr-ucla/direct_lidar_odometry/blob/master/include/nano_gicp/impl/nano_gicp_impl.hpp#L152)

The first is for alignment between adjacent scans in scan-to-scan, and the second is for caching the covariances of keyframe scans internally for future scan-to-map alignments so we can [reuse](https://github.com/vectr-ucla/direct_lidar_odometry/blob/master/src/dlo/odom.cc#L1324) them when building subsequent submaps.

The function `this->gicp_s2s.swapSourceAndTarget()` gets called [_before_](https://github.com/vectr-ucla/direct_lidar_odometry/blob/master/src/dlo/odom.cc#L818) `updateKeyframes()`, which is also after the first time covariances are calculated. As you can probably guess, this [swaps](https://github.com/vectr-ucla/direct_lidar_odometry/blob/master/include/nano_gicp/impl/nano_gicp_impl.hpp#L91) the source and target data structures (cloud, kdtree, and covariances) -- but what's important is that, after this swap, we just care about `this->gicp_s2s`'s target data structures now since, at this point, we've already performed scan-to-scan and don't need the previous scan anymore. So we can just reuse the source structures as temporary storage as indicated by [this comment](https://github.com/vectr-ucla/direct_lidar_odometry/blob/master/src/dlo/odom.cc#L1168).

One might wonder why we're computing covariances twice. Since scan-to-scan and scan-to-map operate in different coordinate systems (`base_link` vs `odom`), we need two sets of covariances for each alignment. One could certainly just transform the first set of covariances using GICP's output, but we found that calculating a single point cloud's covariances took negligible time (versus a variable-sized submap point cloud) and was just easier to implement this way when [fixing this bug](https://github.com/vectr-ucla/direct_lidar_odometry/commit/de9a8c57a9016fc9d78a6cd0bf08d097eab652b9) after initial release (thanks to @RyGuy101).

Let me know if that answers your question.

qweesiic, 2022-05-23:
yes, my question has been answered, thanks

closed 2022-05-23 by qweesiic