vectr-ucla/direct_lidar_odometry #13: update key frames

qweesiic, 2022-05-23:
thanks for fully and quickly answering last question!
now if feature info had been changed largely after mapping ,  the accuracy and stability of  localization must be terrible. 
to avoid this situation,  exactly update key frames whose feature info has changed   should be better solution.
so as your suggestion, how to exactly update key frames whose feature info has changed ? 
thanks.

kennyjchen, 2022-05-24:
Hi @qweesiic -- DLO currently assumes a static environment during runtime and doesn't account for dynamic changes. Empirically speaking though, we've seen promising results in the presence of dynamic objects (e.g., walking people) even without updating keyframe covariances for submap scan-matching, perhaps due to GICP's robustness to incorrect correspondences. Keyframe covariances and map representation could probably be updated continuously after detecting an environmental change somehow though, as you mentioned. Localization in dynamic environments is definitely still an active area of research.

qweesiic, 2022-05-24:
got it! thanks.

closed 2022-05-24 by qweesiic