vectr-ucla/direct_lidar_odometry #14: Map generation

Srichitra-S, 2022-06-15:
Hi,
How is the map generated? What algorithm is used? Can we upload a pcd map of the environment and then apply dlo just for localisation?
I assumed dlo as a localisation algorithm, but it seems to take in the point cloud data from the bag file and generate a map on its own.

kennyjchen, 2022-06-15:
Hi @Srichitra-S -- DLO is an odometry algorithm and generates both pose (6DOF) and map, which is represented by a set of keyed scans throughout the environment. The algorithm currently doesn't support pure localization on a provided map, but this should be pretty straightforward to implement for those who are interested. Note that depending on the map size and complexity, an initial pose may be required. See my comment here for more information: #11

JACKLiuDay, 2022-06-20:
> Hi @Srichitra-S -- DLO is an odometry algorithm and generates both pose (6DOF) and map, which is represented by a set of keyed scans throughout the environment. The algorithm currently doesn't [...]

Hi, thank you for your great work. Does DLO have a map saving function, such as saving in pcd format?

kennyjchen, 2022-06-20:
Hi @JACKLiuDay --

We've added a new service to save DLO's generated map into a `.pcd` in [`v1.4.0`](https://github.com/vectr-ucla/direct_lidar_odometry/releases/tag/v1.4.0) per your request. To use it, please call the following command in a separate terminal window:

```
rosservice call /robot/dlo_map/save_pcd LEAF_SIZE SAVE_PATH
```

JACKLiuDay, 2022-06-21:
> Hi @JACKLiuDay --
>
> We've added a new service to save DLO's generated map into a `.pcd` in [`v1.4.0`](https://github.com/vectr-ucla/direct_lidar_odometry/releases/tag/v1.4.0) per your request. To use [...]

Wow, it's awesome! Thank you for great work! We will test it soon.

HomieRegina, 2022-06-27:
> 

Hi, I have successfully run through the dataset named dlo_test.bag. But I still can't save successfully .pcb file. The following picture is my running result. 
[image]
Could you please tell me how to deal this issue? Looking forward to your reply.

HomieRegina, 2022-06-27:
> Hi @JACKLiuDay --
>
> We've added a new service to save DLO's generated map into a `.pcd` in [`v1.4.0`](https://github.com/vectr-ucla/direct_lidar_odometry/releases/tag/v1.4.0) per your request. To use [...]

By the way, do you have the dataset of the second video in License in README.md? We want to try that dataset. Could you please share it with us?

kennyjchen, 2022-06-27:
Hi @HomieRegina -- don't forget to change the input arguments of the service call, e.g.
```
rosservice call /robot/dlo_map/save_pcd 0.25 ~/Downloads
```

Regarding the multilevel dataset, we'll need to look into if we can publicly share that or not. Thanks.

HomieRegina, 2022-06-28:
> Hi @HomieRegina -- don't forget to change the input arguments of the service call, e.g.
> 
> ```
> rosservice call /robot/dlo_map/save_pcd 0.25 ~/Downloads
> ```
> 
> Regarding the multilevel dataset, we'll need to look into if we can publicly share that or not. Thanks.

Thank you for your reply. But now I have a new problem. When I ran dlo_ test. bag yesterday, rviz can successfully display the map. However, it was not displayed when running today. I tried so many times and the program did not report an error.
[image]
Do you know the reason and how to solve it?

kennyjchen, 2022-06-28:
Might be a topic name mismatch if you modified the launch file. Try double-checking that.

closed 2022-07-01 by kennyjchen