vectr-ucla/direct_lidar_odometry #15: Is this algorithm work for multi-layer enviroment?

JACKLiuDay, 2022-06-17:
Hi, thank you for your brilliant work. I want to know whether DLO suitable for multi-layer structure, such as a big office building.

kennyjchen, 2022-06-17:
Hi @JACKLiuDay -- thanks for your interest. We've seen promising results in multi-level environments (e.g., Figure 9D/E in the paper), but we haven't tested enough for a final conclusion. Empirically though, performance of DLO (and any SLAM algorithm) in multi-level buildings can depend on several variables, including whether the environment has stairs/ramps/something else, how narrow or wide they are, the FOV of your sensor, whether you tilt the sensor or not as you traverse these passageways (i.e., legged vs aerial/handcarries), etc. If you do test in these environments, we'd be interested in hearing about your experiences.

JACKLiuDay, 2022-06-20:
> tilt

Hi，thank you for reply. We are preparing to conduct a SLAM experiment in a coal mining project for robot localization in roadways.  And our group will try testing with DLO, and of course other methods. We will show photos of any experimental progress in the future.

closed 2022-06-20 by JACKLiuDay

Chaoyue-cpu, 2024-11-05:
> > tilt   倾斜
> 
> Hi，thank you for reply. We are preparing to conduct a SLAM experiment in a coal mining project for robot localization in roadways. And our group will try testing with DLO, and of course other methods. We will show photos of any experimental progress in the future.你好，谢谢你的回复。我们正在准备进行一项SLAM实验，旨在为煤矿巷道中的机器人定位提供解决方案。我们的团队将尝试使用DLO，当然也会尝试其他方法。我们会在将来展示任何实验进展的照片。

Have you found an appropriate method?  thanks.