vectr-ucla/direct_lidar_odometry #16: Hardware architecture

HomieRegina, 2022-06-30:
Hi. I have a NVIDIA NX board, which is based on ARM architecture. I encountered the following problems when compiling on the NX board.
file:///media/jhm/KINGSTON/%E5%BE%AE%E4%BF%A1%E5%9B%BE%E7%89%87_20220630112334.jpg
Can this package run under arm architecture? Is this error related to the architecture? How to solve this issue?

kennyjchen, 2022-06-30:
Hi @HomieRegina -- yes, this was something we also saw during our internal tests on a Raspberry Pi (also ARM). ARM-based boards apparently do not have the `cpuid.h` system header file.

We've released a patch in [`v1.4.1`](https://github.com/vectr-ucla/direct_lidar_odometry/releases/tag/v1.4.1) which should fix this.

HomieRegina, 2022-07-01:
> Hi @HomieRegina -- yes, this was something we also saw during our internal tests on a Raspberry Pi (also ARM). ARM-based boards apparently do not have the `cpuid.h` system header file.
>
> We've [...]

Thanks for your great work. We will test it soon.

closed 2022-07-01 by kennyjchen