vectr-ucla/direct_lidar_odometry #17: HI, I use the nanoicp to location, but it cann`t work

tust13018211, 2022-07-01:
Thanks for your  work, I want to use nanoicp to location, it cann`t work right, but I use the pcl Gicp， it can work,  so, there is anything i need to do?

tust13018211, 2022-07-01:
I found that the translation grows fast，it seems strangeness

kennyjchen, 2022-07-01:
Hi @tust13018211 -- are you using NanoGICP as part of DLO or outside of this package? Thanks.

tust13018211, 2022-07-02:
I  try  to use in dlo and outside , all the result is not good

tust13018211, 2022-07-02:
[image]
[image]

kennyjchen, 2022-07-02:
Are covariances and/or kdtrees of both clouds being properly set in your code?

tust13018211, 2022-07-03:
Convariances is ok， but kdtree  i do not konwn how to do properly for map .  i want to know kdtree and convariance is both need,  and how to set this for map

tust13018211, 2022-07-03:
sorry, I go to see the aticle, I found the Convariances of map use the keyframe`s Convariances to approximated, but now I have the map generate by load or other, so how can set the map Convariances when I do the location using nanogicp

tust13018211, 2022-07-03:
at first, I use the
m_iGicpScanToMap_.setInputTarget(map);
m_iGicpScanToMap_.calculateTargetCovariances();
to built target cloude Covariances and kdtree, and then
m_iGicpScanToMap_.setInputSource(m_pCurrentScan_);
m_iGicpScanToMap_.calculateSourceCovariance
use this to built source cloude Covariances and kdtree, at last,
m_iGicpScanToMap_.align(*aligned, init_pose.matrix.cast());

is there any pronlem above ?

kennyjchen, 2022-07-05:
Hmm looks fine to me. Just to clarify: does DLO work on your end (without any modifications)? And in your code above, switching out NanoGICP with PCL's GICP works? The package should work as a replacement for PCL's GICP (sans some of the computational benefits from data structure recycling). A bit hard to help debug without more information.

tust13018211, 2022-07-06:
Thaks your reply, I just want to test nanogicp for location with a map, I adjust the parameter, it seems work. the result in the picture,  the result seems not very good 
[image]

kennyjchen, 2022-07-06:
Oh I see. Unfortunately we have not tested DLO (or NanoGICP) for localization with a provided map (as mentioned [here](https://github.com/vectr-ucla/direct_lidar_odometry/issues/14) and [here](https://github.com/vectr-ucla/direct_lidar_odometry/issues/11)) and assumes no prior knowledge of the environment, so your results may vary.

closed 2022-07-08 by kennyjchen