vectr-ucla/direct_lidar_odometry #20: when i run "catkin build",it faild and the error logs is following:

NUDTbowen, 2022-07-18:
[ 88%] Linking CXX executable /home/swc/study/slam_study/catkin_ws/devel/lib/direct_lidar_odometry/dlo_odom_node
/usr/bin/ld: warning: libboost_thread.so.1.65.1, needed by /usr/lib/x86_64-linux-gnu/libpcl_common.so, may conflict with libboost_thread.so.1.57.0
/usr/bin/ld: warning: libboost_chrono.so.1.65.1, needed by /opt/ros/melodic/lib/libroscpp.so, may conflict with libboost_chrono.so.1.57.0
/usr/bin/ld: warning: libboost_filesystem.so.1.65.1, needed by /opt/ros/melodic/lib/libroscpp.so, may conflict with libboost_filesystem.so.1.57.0
/usr/bin/ld: warning: libboost_regex.so.1.65.1, needed by /opt/ros/melodic/lib/librosconsole.so, may conflict with libboost_regex.so.1.57.0
/usr/bin/ld: warning: libboost_system.so.1.57.0, needed by /home/swc/anaconda3/envs/badgr/lib/libboost_filesystem.so, may conflict with libboost_system.so.1.65.1
/usr/bin/ld: warning: libicuuc.so.54, needed by /home/swc/anaconda3/envs/badgr/lib/libboost_regex.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libicui18n.so.54, needed by /home/swc/anaconda3/envs/badgr/lib/libboost_regex.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libicudata.so.54, needed by /home/swc/anaconda3/envs/badgr/lib/libboost_regex.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libpcl_filters.so.1.9, needed by /home/swc/anaconda3/envs/badgr/lib/libpcl_segmentation.so, may conflict with libpcl_filters.so.1.8
/usr/bin/ld: warning: libpcl_search.so.1.9, needed by /home/swc/anaconda3/envs/badgr/lib/libpcl_segmentation.so, may conflict with libpcl_search.so.1.8
/usr/bin/ld: warning: libpcl_common.so.1.9, needed by /home/swc/anaconda3/envs/badgr/lib/libpcl_segmentation.so, may conflict with libpcl_common.so.1.8
/usr/bin/ld: warning: libboost_filesystem.so.1.72.0, needed by /home/swc/anaconda3/envs/badgr/lib/libpcl_recognition.so, may conflict with libboost_filesystem.so.1.57.0
/usr/bin/ld: warning: libboost_thread.so.1.72.0, needed by /home/swc/anaconda3/envs/badgr/lib/libpcl_io.so, may conflict with libboost_thread.so.1.57.0
/usr/bin/ld: warning: libboost_iostreams.so.1.72.0, needed by /home/swc/anaconda3/envs/badgr/lib/libpcl_io.so, may conflict with libboost_iostreams.so.1.57.0
/home/swc/anaconda3/envs/badgr/lib/libboost_regex.so: undefined reference to `u_tolower_54'
/home/swc/anaconda3/envs/badgr/lib/libboost_regex.so: undefined reference to `icu_54::Locale::Locale(icu_54::Locale const&)'
/home/swc/anaconda3/envs/badgr/lib/libboost_regex.so: undefined reference to `icu_54::Locale::Locale()'
/home/swc/anaconda3/envs/badgr/lib/libboost_regex.so: undefined reference to `icu_54::Locale::~Locale()'
/home/swc/anaconda3/envs/badgr/lib/libboost_regex.so: undefined reference to `u_charFromName_54'
/home/swc/anaconda3/envs/badgr/lib/libboost_regex.so: undefined reference to `u_isblank_54'
/home/swc/anaconda3/envs/badgr/lib/libboost_regex.so: undefined reference to `icu_54::Collator::createInstance(icu_54::Locale const&, UErrorCode&)'
/home/swc/anaconda3/envs/badgr/lib/libboost_regex.so: undefined reference to `u_charType_54'
/home/swc/anaconda3/envs/badgr/lib/libboost_regex.so: undefined reference to `u_digit_54'
/home/swc/anaconda3/envs/badgr/lib/libboost_regex.so: undefined reference to `u_isspace_54'
collect2: error: ld returned 1 exit status
direct_lidar_odometry/CMakeFiles/dlo_odom_node.dir/build.make:364: recipe for target '/home/swc/study/slam_study/catkin_ws/devel/lib/direct_lidar_odometry/dlo_odom_node' failed
make[2]: *** [/home/swc/study/slam_study/catkin_ws/devel/lib/direct_lidar_odometry/dlo_odom_node] Error 1
CMakeFiles/Makefile2:666: recipe for target 'direct_lidar_odometry/CMakeFiles/dlo_odom_node.dir/all' failed
make[1]: *** [direct_lidar_odometry/CMakeFiles/dlo_odom_node.dir/all] Error 2
Makefile:145: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

kennyjchen, 2022-07-18:
Looks like it's not finding `libicu`, probably since you're in a conda environment and installed Boost that way. My first tip would be to uninstall conda. My second tip would be to try `sudo apt install libicu-dev` (or the equivalent in conda if you don't follow my first tip).

NUDTbowen, 2022-07-18:
您的邮件我已收到，请放心。

closed 2022-07-21 by kennyjchen