vectr-ucla/direct_lidar_odometry #21: Conversion of lidar and IMU coordinate system

HomieRegina, 2022-08-17:
Hi! I use a vlp16 lidar and a microstran imu(3DM-GX5-25) to record bags. And here is the part of the map I built with your algorithm. Because the coordinate systems of IMU and lidar are not consistent in the physical direction, this has affected the construction of the two-layer environment. Could you please tell me how to convert IMU and lidar coordinate system?
[image]

kennyjchen, 2022-08-18:
DLO currently assumes that IMU and LiDAR coordinate systems roughly coincide. I may add support for extrinsics into DLO in the future, but for now try mounting your IMU upside-down so that it points forward-left-up (ROS convention) if possible. It's a bit hacky but it should work.

That being said, can you point out the issue in your map? It's a little hard to tell what is going on since all points are colored red. Thanks.

HomieRegina, 2022-08-18:
Thank you for your reply. I will try your method again. 
About your question: this picture is the part of the map I built. The picture below is the whole map. This is the result of adjusting the parameters. The map is tilted as a whole. But the two floors are relatively parallel.  Do you have any good adjustment suggestions for this situation?
[image]

kennyjchen, 2022-08-22:
If the floors are relatively parallel (which they look like they are), but the whole map is tilted, it could be a bad initial position as we discussed [here](https://github.com/vectr-ucla/direct_lidar_odometry/issues/18). Try starting from a level surface.

HomieRegina, 2022-08-23:
Thank you for your reply and I will try again. Thanks for your good job and patience.

closed 2022-08-23 by kennyjchen