vectr-ucla/direct_lidar_odometry #22: How to correct point cloud caused by motion？

JACKLiuDay, 2022-08-17:
Hi, thank you for your great work about DLO. I have a question. When the robot is moving fast, the lidar point cloud will generate distortion due to movement, which will cause negative impact on the construction of the map. So how did DLO deal with this problem?

kennyjchen, 2022-08-18:
DLO does not correct for motion distortion and assumes that either the scan is deskewed by an upstream module, or that the effect is negligible due to low movement speeds or a high sensor rate.

JACKLiuDay, 2022-08-19:
> DLO does not correct for motion distortion and assumes that either the scan is deskewed by an upstream module, or that the effect is negligible due to low movement speeds or a high sensor rate.

Thank you for your reply.

closed 2022-08-19 by JACKLiuDay