vectr-ucla/direct_lidar_odometry #24: I feel a littl strange about the parameter  kCorrespondences with scan to scan ?

tust13018211, 2023-01-15:
Thanks for your work, I test the DLO in my data, I find set parameter  kCorrespondences=1 with scan to scan is  better than kCorrespondences=10, In my opinon, kCorrespondences=1 means gicp is the same with icp, so I donn`t konw the reason why?

kennyjchen, 2023-01-17:
Hi @tust13018211 -- 

Depending on your sensor setup, it could be that your point cloud is sparse enough that having less `kCorrespondences` provides a more accurate scan-to-scan solution (since computed surface covariances might be inaccurate), thus providing scan-to-map with an inaccurate prior (since scan-to-scan is only used as a prior to scan-to-map in DLO).

closed 2023-01-25 by kennyjchen

tust13018211, 2023-01-30:
sorry replay latter. In fact, my data is warehouse and lidar Towards the ceiling. So I think there is enough surface.  so the result a little strange.