vectr-ucla/direct_lidar_odometry #29: Bug in adaptive threshold.

Gatsby23, 2023-04-14:
Thank you for your wonderful work, but it seems there is a little bug in the adaptive threshold estimation:
From the paper, we know  "Thus, we choose to scale the translational threshold for new keyframes according to the “spaciousness” in the instantaneous point cloud scan, defined as $m_k = α m_{k−1} + β M_k$, where M_k is the median Euclidean point distance from the origin to each point in the preprocessed point cloud, α = 0.95, β = 0.05,". However, the code in odom.cc for calculation "spaciousness" is:
```
  std::nth_element(ds.begin(), ds.begin() + ds.size()/2, ds.end());
  float median_curr = ds[ds.size()/2];
  static float median_prev = median_curr;
  float median_lpf = 0.95*median_prev + 0.05*median_curr;
  median_prev = median_lpf;
``` 
It seems it always uses the median distance as "spaciousness".

kennyjchen, 2023-04-14:
Hi @Gatsby23 -- thanks for your comment.

So since we declared line 3 as static, it only gets called on the first iteration in order to initialize the value for line 4 the first time a scan is received. So the first value of `median_lpf` will equal to the median distance. However, in subsequent iterations, line 5 calculates `median_prev` instead, creating a simple low pass filter. If you print out the values of `median_curr` (median distance) and `median_lpf`, you'll see that they're different values after the first iteration. Let me know if that helps.

Gatsby23, 2023-04-16:
> Hi @Gatsby23 -- thanks for your comment.
>
> So since we declared line 3 as static, it only gets called on the first iteration in order to initialize the value for line 4 the first time a scan is [...]

OK， It's my mis-understood. Thank you for your comment! I really like this work: simple and effective.

closed 2023-04-16 by Gatsby23