vectr-ucla/direct_lidar_odometry #33: Initialization function

numb0824, 2023-06-06:
Hi，
     Thanks for your great work！ I have been testing 'DLO' for a long time and its performance is excellent .
     In some tests, due to environmental reasons such as a particularly sparse environment, DLO results are incorrect. Of course, this is not a problem for 'DLO' itself. When this situation occurs, I don't want to shut down DLO and restart it, but rather want to initialize it, just like the initialpose function in ROS.
    So could you please add features like "initialpose" to 'DLO' to solve the problem of not turning off 'DLO' and continuing to run in case of inaccurate pose caused by accidents?
    I apologize to you if my request is very presumptuous.
     Thank you again for your great work .

Best wishes

kennyjchen, 2023-06-07:
Hi @numb0824 -- thanks for using our work. Do you mean just re-initializing the pose? Or perhaps a complete reset of DLO?

kennyjchen, 2023-06-13:
Hmm... I'll have to think about this. It's a bit tricky since DLO has separate nodes for odometry and mapping. In the meanwhile, we welcome any PR's to add this functionality!