vectr-ucla/direct_lidar_odometry #34: large drift in z-axis

WeilongXia, 2023-06-16:
Thanks a lot for open sourcing such a good job! But I found during the experiment that the odometry drifts a lot in the z-axis estimation. I walked about 600m with the Lidar in hand, and when I returned to the starting point, I could see that the created point cloud map would gradually become warped. Is there any good way to solve it, such as parameter adjustment or exercise mode change? The Lidar I use is Livox MID360. Looking forward to hearing from you.
[image]
[image]

kennyjchen, 2023-06-16:
Hi @WeilongXia -- check out [this issue.](https://github.com/vectr-ucla/direct_lidar_odometry/issues/18).

If you are not using gravity alignment, DLO's reference frame depends on its initial orientation (so if the initial orientation is tilted, the entire map will be tilted). Try starting DLO with your sensor on a flat surface. If you are using gravity alignment, make sure your LiDAR and IMU are horizontally-aligned otherwise it will not work (and that your IMU is sufficiently high-quality). Let me know if that helps.

WeilongXia, 2023-06-17:
Hi, I have tried placing the Lidar horizontally when the program is on, and the effect has been greatly improved! Thanks for you help!
[image]
[image]

closed 2023-06-17 by WeilongXia