vectr-ucla/direct_lidar_odometry #36: How to improve the publish frequency of odom?

Joosoo1, 2023-06-20:
Hello great author, the point cloud publish frequency of my lidar is 10hz, at this time the frequency of odom output from DLO is also 10hz, is there any way to increase the publish frequency of odom, I have some work that needs high frequency pose estimation.thanks！

Joosoo1, 2023-06-20:
This is a very good job, thank you very much for making it open source!

kennyjchen, 2023-06-21:
Since DLO is LiDAR-only, the update rate is limited by how fast new measurements are received. You could artificially increase the pose estimate by putting the publisher on a ros::Timer, but that will just republish the same pose multiple times until a new update comes in.

You can try using DLIO (I see that you are already), so that the IMU helps with faster state estimation.

Joosoo1, 2023-06-23:
Okay, I get it, thanks your reply!

closed 2023-06-23 by Joosoo1