vectr-ucla/direct_lidar_odometry #37: About IMU Preintegration!

yuchuxiang, 2023-07-06:
Hi authors, I read your code and in the pre-integration part, IMU is used to estimate the rotation only, but not the translation. Is it because your application scenario is a low-speed environment and does not require translation prediction?

kennyjchen, 2023-07-11:
Hi @yuchuxiang -- thanks for the question. Yes, DLO is loosely-coupled and only uses IMU to compute a rotational prior for scan-matching. The [successor](https://github.com/vectr-ucla/direct_lidar_inertial_odometry) to DLO uses IMU for both a rotational and translational prior, so if you're interested or your application requires it, check that out!

yuchuxiang, 2023-07-14:
> Hi @yuchuxiang -- thanks for the question. Yes, DLO is loosely-coupled and only uses IMU to compute a rotational prior for scan-matching. The [...]

Thank you very much for your excellent work!

closed 2023-07-14 by yuchuxiang