vectr-ucla/direct_lidar_odometry #41: Loop closure

wienans, 2023-08-18:
Hey this is more of a question than a issue.
I wanted to ask if your system does any kind of Loop closure to make sure that for big maps where you revisit a certain location after a time the drift of the map is minimal.

As in the videos or GIFs we never see a "drifted" map which gets visually corrected, but all maps look very precise i was wondering if the algorithms drift is that small or if there is some loop closure happening.

Greetings and thanks for the great work
Sven

kennyjchen, 2023-08-31:
Hi @wienans -- DLO is purely an odometry algorithm and doesn't contain any explicit loop closures for retroactive map correction.

wienans, 2023-08-31:
Okay thanks for info :)

closed 2023-09-01 by kennyjchen