vectr-ucla/direct_lidar_odometry #44: Drift using 2D Laser in large spaces

LotfiZ, 2023-11-01:
Hi dear,

its been few weeks that we are testing DLO to compute odometry for a mobile robot and its has great result. However, when we went to a larger space to test it, we had a lot of drift, we did some parameter tuning as suggested in the previous issue tickets concerning sparse lidars but we could not reach acceptable results. The laser measurements are in 2D and do not cover full 360 but around 270, in addition, sometimes the laser measures the ground. Do you think that its related  to this ?  

[image]

We played with the outlier rejection parameters, the max corr distance for both s2s and s2m and the adaptiveParams. An error is raised everytime in this environnement and its related to convexHull : `[pcl::ConvexHull::performReconstrution3D] ERROR: qhull was unable to compute a convex hull for the given point cloud (375)!`. This error does not rise in smaller places.

Thank you very much for your help !

kennyjchen, 2024-05-16:
Yes, a non-360 FOV can definitely affect how submapping is done in DLO.

aba01kz, 2024-08-22:
sorry, can I use DLO with rplidar (2D lidar with single line laser)?