vectr-ucla/direct_lidar_odometry #50: Question about convex hull of the keyframe set

Wuqiqi123, 2024-05-27:
[image]
Hi, author. Thanks for you great works.

I have a question about  convex hull in your paper:
As the image above, the convex hull of the keyframe set are the L nearest neighbor convex hull scans of all keyframes, why the left-up corner keyframes are been selected?

kennyjchen, 2024-10-03:
Hi @Wuqiqi123 -- great question. Those top-left ones are keyframes that are in the set of all keyframes which construct the convex hull of the keyframes. We use those to retrieve boundary points for the submap during scan-to-map matching.

Wuqiqi123, 2024-10-08:
> Hi @Wuqiqi123 -- great question. Those top-left ones are keyframes that are in the set of all keyframes which construct the convex hull of the keyframes. We use those to retrieve boundary points for the submap during scan-to-map matching.

Given a set of all keyframes(coordinate system) and to choose some of them to construct the convex hull, but I don't known why only the green spheres are selected. The frames with green spheres doesn't look like a convex hull boundary of all keyframes.